devinaconley/arduino-plotter #3: some bugs in the listener, and some refactoring

d-day, 2016-10-07:
Hello Devin:

I have been hacking on your plotter software toward using the graph to display information from an old but high-precision digital multimeter.

I noticed the large update yesterday and decided to share some of my code just to make you aware that I'm working on it.

Because I wouldn't want you to try and merge any of this code yet, I have saved what I have as [a gist, here](https://gist.github.com/d-day/b6d96a9a7a27080702ee0ff149ae2951).

I am sure some of the stuff in there will meet with annoyance (the spacing, and the debugging lines). And other stuff will meet with a better response (the refactoring attempts). There is a lot of potential for refactoring yet, so I'll keep you updated.

Thanks,
Drew

devinconley, 2016-10-07:
Thanks for reaching out Drew - just went through the gist and commented

d-day, 2016-10-14:
I updated the gist and commented. I finally got to the point where I did refactoring and fixed errors I found, and eliminated all of the bugs I introduced. At this point, I can start adding the features I need or go elsewhere.

Do you have a way to be contacted via a more realtime method? IRC?

devinaconley, 2017-01-05:
Will be working through some big changes to the listener in the branch named "listener-overhaul"

devinaconley, 2017-01-27:
Listener has been completely rewritten so i'm closing this - feel free to open a new issue if you see any more problems.

closed 2017-01-27 by devinaconley