devinaconley/arduino-plotter #4: Enhancement request

Perry-RU, 2016-11-23:
Support for 7+ variable graphs.
Support to choose line color on graph of specified variable.

devinaconley, 2016-11-26:
Will look at some possibilities for 7+ variable graphs. I wanted to avoid arbitrary length argument lists for safety, but that very well might end up being the best option.

Planning on overhauling the listener soon anyways -- will include configurable line color in that effort

devinaconley, 2017-02-07:
Configurable color was added in latest release (#5). Support for 7+ variable graphs should be in soon

phdv61, 2019-06-06:
Thank you for sharing. 
Instead of having dotted lines being plotted, I prefer continuous lines. 
I modifed this section your code accordingly as follows :

   // Do actual data plotting
  
  // 1 - Is the rolling buffer full of data ?
  
  if ( this.currPoints < this.maxPoints )  //   Not yet

	for ( int i = 0; i < this.numVars; i++ )
	{
	    this.parent.stroke( this.colors[i] );
	    for ( int j = 0; j < this.currPoints-1; j++ )
	    {
  		this.parent.line( (float)(this.posX + (this.data[j][i][0]*xScale - xOffset)), (float)(this.posY + yOffset - data[j][i][1]*yScale),
                        (float)(this.posX + (this.data[j+1][i][0]*xScale - xOffset)), (float)(this.posY + yOffset - data[j+1][i][1]*yScale) );
	    }
	}	
	
  else  // 2   YES, roll-back did start continuous plotting must be done in two halves
  
  {  
   for ( int i = 0; i < this.numVars; i++ )
   {
      this.parent.stroke( this.colors[i] );
      for ( int j = this.index ; j < this.maxPoints-1; j++ )
      {
      this.parent.line( (float)(this.posX + (this.data[j][i][0]*xScale - xOffset)), (float)(this.posY + yOffset - data[j][i][1]*yScale),
                        (float)(this.posX + (this.data[j+1][i][0]*xScale - xOffset)), (float)(this.posY + yOffset - data[j+1][i][1]*yScale) );
      }
   }      
     for ( int i = 0; i < this.numVars; i++ )
   {
      this.parent.stroke( this.colors[i] );
      for ( int j = 0; j < this.index-1; j++ )
      {
      this.parent.line( (float)(this.posX + (this.data[j][i][0]*xScale - xOffset)), (float)(this.posY + yOffset - data[j][i][1]*yScale),
                        (float)(this.posX + (this.data[j+1][i][0]*xScale - xOffset)), (float)(this.posY + yOffset - data[j+1][i][1]*yScale) );
      }
   }
   
  }