devinaconley/arduino-plotter #6: Listener Configuration Error

Jack-NZ, 2017-02-04:
When the listener tries to configure itself, it does not increment portIndex, and does not proceed beyond Port 0.

devinaconley, 2017-02-04:
What OS are you on? And are you running an example or a personal script?

Jack-NZ, 2017-02-04:
Windows 10, and I'm running the example listener script for an Arduino Mega 2560 on COM4

devinaconley, 2017-02-05:
Okay and what script is running on the Arduino?

Jack-NZ, 2017-02-05:
The quick_start script provided with the installed library.

devinaconley, 2017-02-07:
Just tested on Windows 10 and was not able to replicate the issue. Are you making sure that the serial monitor is closed in the Arduino IDE?

Jack-NZ, 2017-02-09:
I'm positive it's closed. There are no functions to start the serial monitor in the Arduino script, and I have no external scripts running that would interfere with the serial communication.

devinaconley, 2017-02-09:
Try changing 
```arduino
x = 10sin( 2.0PI*( millis() / 5000.0 ) );
```
to
```arduino
x = 10*sin( 2.0*PI*( millis() / 5000.0 ) );
```

Jack-NZ, 2017-02-09:
Done, still no response from the listener.

devinaconley, 2017-02-09:
Hmm very weird. Can you copy and paste a sample of what gets printed when you do open the serial monitor? You'll have to switch the baud rate to 115200 to get anything meaningful.

Also have you tried running the stand-alone listener? ( [Windows 64-bit](https://www.dropbox.com/s/0x4xqv375h4j5hh/arduino-plotter-listener-windows64.zip?dl=1)  ) Or only the script from within the Processing IDE?

Jack-NZ, 2017-02-09:
The serial monitor repeats at 115200 baud:
```
Some title of a graph@0@500@1@label for x@green@-9.8846235@#
#1@1@500@0@
```
And a series of random characters at 9600 baud. 

I've tried running the IDE script and the external listener, but neither have worked.

devinaconley, 2017-02-09:
Okay - I just pushed a few changes to the source listener script and the Graph.java class for better logging. Can you re-download both of those and run it from the processing IDE?

Then can you copy and paste any log messages from the console?

vcortex91, 2017-02-10:
I just downloaded this -- same issue.  Windows 7, 64-bit.  Guessing it's the startup routine in listener.java.

devinaconley, 2017-02-10:
Have you tried running the listener with the Processing IDE? If so can you copy and paste the console output here?

Still have not been able to recreate this issue on my machine

vcortex91, 2017-02-10:
Yes, I just tried this and same result.  In listener.java, I tried changing initial portIndex (line 93) to my port (4), but it ends up getting set back to port 0 due to passing condition if ( portIndex >= Serial.list().length ) on line 119.  I then tried to change the port value inside the if statement to "4" as well, but it still just sits there scanning for Port 4.

vcortex91, 2017-02-10:
I tried another Processing sketch that pulls from Arduino COM port and it worked fine.  So seems to be some kind of issue with Java.

devinaconley, 2017-02-10:
Is COM4 the same port that you were able to successfully connect with in the other sketch?

devinaconley, 2017-02-10:
Alright - think I found the issue actually. Will put in a fix within the hour

devinaconley, 2017-02-10:
Okay... Looks like the hang was occurring when one of the serial ports was valid but busy (specifically bluetooth for me when I recreated it) and the Serial constructor wasn't timing out in a reasonable amount of time.

I nested the connecting functionality within an ExecutorService and Future to force a timeout at 2 seconds.

Let me know if the script works now and if so I will export a new set of standalone listeners

vcortex91, 2017-02-11:
Makes it farther this time, but still seems to be an issue.  This is what the Arduino is outputting (from quick_start sketch):

Some title of a graph@0@500@1@label for x@6.11@#
#1@1@500@0@

And this is the console output:
Attempting connect on port: COM4
Connected on COM4. Listening for configuration...
Attempting connect on port: COM4
Attempting connect on port: COM4
Attempting connect on port: COM4
Attempting connect on port: COM4
Attempting connect on port: COM4
Attempting connect on port: COM4

vcortex91, 2017-02-11:
With DEBUG turned on:

Attempting connect on port: COM4
Connected on COM4. Listening for configuration...
Exception... null
Exception... null
...
Exception... null
Exception... null
Trying next port... index: 0, name: COM4
Attempting connect on port: COM4
Exception... null
java.lang.RuntimeException: Error opening serial port COM4: Port busy
Exception... null
Exception... null
Trying next port... index: 0, name: COM4
Exception... null
Attempting connect on port: COM4
java.lang.RuntimeException: Error opening serial port COM4: Port busy
Exception... null
Exception... null
Exception... null
Trying next port... index: 0, name: COM4
Attempting connect on port: COM4
java.lang.RuntimeException: Error opening serial port COM4: Port busy
Exception... null
Exception... null

vcortex91, 2017-02-11:
If you don't want to investigate this further, I understand.

devinaconley, 2017-02-11:
So it looks the listener did successfully connect at the beginning. Is the Arduino running quick_start on COM4?

devinaconley, 2017-02-11:
Happy to help with getting this figured out

vcortex91, 2017-02-11:
Yes it is

devinaconley, 2017-02-11:
Try out my latest commit. Added some legitimate logging

vcortex91, 2017-02-11:
[Time: 330 ms][STATUS] Attempting connect on port: COM4
[Time: 352 ms][STATUS] Connected on COM4. Listening for configuration...
[Time: 5357 ms][STATUS] Attempting connect on port: COM4
[Time: 5367 ms][STATUS] Attempting connect on port: COM4
[Time: 5371 ms][STATUS] Connected on COM4. Listening for configuration...
[Time: 10375 ms][STATUS] Attempting connect on port: COM4
[Time: 10385 ms][STATUS] Attempting connect on port: COM4
[Time: 10390 ms][STATUS] Connected on COM4. Listening for configuration...

devinaconley, 2017-02-12:
Can you run in debug mode?

Also have you checked your serial monitor output in the Arduino IDE? See [this comment](https://github.com/devinaconley/arduino-plotter/issues/6#issuecomment-278522885) for what it should look like

vcortex91, 2017-02-12:
Yes, running in Debug mode gives:
[Time: 234 ms][STATUS] Attempting connect on port: COM4
[Time: 265 ms][DEBUG] Exception on connect: java.util.concurrent.ExecutionException: java.lang.RuntimeException: Error opening serial port COM4: Port not found
[Time: 5244 ms][DEBUG] Trying next port... index: 0, name: COM4
[Time: 5244 ms][STATUS] Attempting connect on port: COM4
[Time: 5244 ms][DEBUG] Exception on connect: java.util.concurrent.ExecutionException: java.lang.RuntimeException: Error opening serial port COM4: Port not found
[Time: 5244 ms][DEBUG] Trying next port... index: 0, name: COM4
[Time: 5244 ms][STATUS] Attempting connect on port: COM4
[Time: 5244 ms][DEBUG] Exception on connect: java.util.concurrent.ExecutionException: java.lang.RuntimeException: Error opening serial port COM4: Port not found
[Time: 5260 ms][DEBUG] Trying next port... index: 0, name: COM4
[Time: 5261 ms][STATUS] Attempting connect on port: COM4
[Time: 5262 ms][DEBUG] Exception on connect: java.util.concurrent.ExecutionException: java.lang.RuntimeException: Error opening serial port COM4: Port not found
[Time: 5278 ms][DEBUG] Trying next port... index: 0, name: COM4
[Time: 5278 ms][STATUS] Attempting connect on port: COM4
[Time: 5278 ms][DEBUG] Exception on connect: java.util.concurrent.ExecutionException: java.lang.RuntimeException: Error opening serial port COM4: Port not found
[Time: 5278 ms][DEBUG] Trying next port... index: 0, name: COM4
[Time: 5278 ms][STATUS] Attempting connect on port: COM4
[Time: 5278 ms][DEBUG] Exception on connect: java.util.concurrent.ExecutionException: java.lang.RuntimeException: Error opening serial port COM4: Port not found
[Time: 5293 ms][DEBUG] Trying next port... index: 0, name: COM4
[Time: 5294 ms][STATUS] Attempting connect on port: COM4
[Time: 5294 ms][DEBUG] Exception on connect: java.util.concurrent.ExecutionException: java.lang.RuntimeException: Error opening serial port COM4: Port not found
[Time: 5311 ms][DEBUG] Trying next port... index: 0, name: COM4
[Time: 5311 ms][STATUS] Attempting connect on port: COM4
[Time: 5311 ms][DEBUG] Exception on connect: java.util.concurrent.ExecutionException: java.lang.RuntimeException: Error opening serial port COM4: Port not found
[Time: 5311 ms][DEBUG] Trying next port... index: 0, name: COM4
[Time: 5311 ms][STATUS] Attempting connect on port: COM4
[Time: 5311 ms][DEBUG] Exception on connect: java.util.concurrent.ExecutionException: java.lang.RuntimeException: Error opening serial port COM4: Port not found
[Time: 5326 ms][DEBUG] Trying next port... index: 0, name: COM4
[Time: 5327 ms][STATUS] Attempting connect on port: COM4
[Time: 5327 ms][DEBUG] Exception on connect: java.util.concurrent.ExecutionException: java.lang.RuntimeException: Error opening serial port COM4: Port not found
[Time: 5344 ms][DEBUG] Trying next port... index: 0, name: COM4
[Time: 5344 ms][STATUS] Attempting connect on port: COM4
[Time: 5344 ms][DEBUG] Exception on connect: java.util.concurrent.ExecutionException: java.lang.RuntimeException: Error opening serial port COM4: Port not found

And yes, the Arduino Serial monitor output is:
Some title of a graph@0@500@1@label for x@7.77@#
#1@1@500@0@

vcortex91, 2017-02-12:
I'm going to have to stop here, as I've found an alternative solution and have moved on to other things

devinaconley, 2017-02-12:
Okay - sounds good. But for reference, looks like you are running an old version of the plotter library on the arduino side, which is why the listener will fail to configure. Newest is v2.1.0

vcortex91, 2017-02-12:
Okay, I missed that.  Arduino library manager stops at 2.0.0.  Looks like it works now

vcortex91, 2017-02-12:
Also, looks like it works with the original listener.exe application, so I'm wondering if that was the problem the whole time

devinaconley, 2017-02-12:
Okay - good to hear. That likely was the problem from the beginning then, might need to update your library manager's indexing because 2.1.0 should be in there.

Hopefully this hardening on the listener side fixed @Jack-NZ 's issue

devinaconley, 2017-02-16:
@Jack-NZ 

Closing this - assuming your problem is fixed unless I hear otherwise

closed 2017-02-16 by devinaconley