devinaconley/arduino-plotter #7: listener: OS X El Capitan complains about damaged file

idank, 2017-02-11:
Looks like the precompiled binary for OS X doesn't work on some versions? When opened I get a dialog saying the file is bad and it moves it to the trash.

Specifically I'm running 10.11.6.

devinaconley, 2017-02-11:
Okay thanks for letting me know - will try to figure that out. At the very least I need to document which versions are supported.

In the meantime, you can run the listener through the Processing IDE. Further, if you export the application from there, it will definitely work on your own system.

devinaconley, 2017-03-23:
Looks like this is an issue with Gatekeeper, the new code signing feature for Mac. Seems to be causing problems for all exported Processing applications.

You can either changes security settings to allow applications from all developers, or run/export the listener from Processing on your own system.

devinaconley, 2017-03-25:
Probably the best option actually: in terminal, run ```xattr -rc /path/to/listener.app```

closed 2017-03-25 by devinaconley