devinaconley/arduino-plotter #8: Document serialization format

idank, 2017-02-13:
Hey, it would be awesome if there was a short wiki page on the serialization format so one could write a different frontend (e.g. browser based). Thanks for your work!

devinaconley, 2017-02-13:
I've actually been meaning to convert to using flattened JSON for serialization to be more robust, support future features, etc. (Probably should have been doing that from the beginning). Will make that change in the next few days and document it as a wiki page. 

If you do write a browser based listener, would be awesome to get that merged.

idank, 2017-02-13:
Sweet, thanks.

I'd be worried about the json encoding overhead (space & time), so might want to profile it before you make the switch.

devinaconley, 2017-02-13:
Thanks, will do that. Planning on writing JSON manually on the Arduino side, so only worried about the cost of parsing it on the listener side.

devinaconley, 2017-02-16:
Did some time profiling and it turned out that the time spent decoding JSON is almost negligible relative to serial transfer costs. Even though a full JSON serialization is more expensive in terms of size, it made it extremely easy to identify different packet formats, meaning I can cut back to sending the full listener configuration at an interval of say 100 packets. Ultimately, I ended up improving the data transfer speed by over 2x.

For reference/record...

Benchmarking on a 5 variable graph vs. time with 1000 points displayed per variable. Timed how long it took to plot 1000 times ( so transferring 5000 values and reloading the entire graph  )

Original serialization
- Total: 14400ms
- Time spent parsing packets: 12ms

Continuous JSON serialization
- Total: 20300ms
- Time spent parsing packets: 41ms

Interval JSON serialization
- Total: 7100ms
- Time spent parsing packets: 26ms

devinaconley, 2017-02-16:
https://github.com/devinaconley/arduino-plotter/wiki/Serialization-Format

Closing this - let me know if you have any issues

closed 2017-02-16 by devinaconley