devinaconley/arduino-plotter #13: Compile Error in Arduino Due

shouryasingh, 2017-03-25:
I am Getting following error while compiling program in Arduino Interface for Arduino Due Board  :-
-----------------------------------------------------------------------------------------------------------
Arduino: 1.8.0 (Windows 7), Board: "Arduino Due (Programming Port)"

C:\Users\Shourya\Documents\Arduino\libraries\Plotter\Plotter.cpp: In member function 'void Plotter::Graph::Plot(bool)':

C:\Users\Shourya\Documents\Arduino\libraries\Plotter\Plotter.cpp:244:45: error: 'dtostrf' was not declared in this scope

  dtostrf( wrappers[i].GetValue(), 1, 7, val );
                                            ^

exit status 1
Error compiling for board Arduino Due (Programming Port).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
-----------------------------------------------------------------------------------------------------------

Please look into the matter.

devinaconley, 2017-03-25:
Try adding ```#include <avr/dtostrf.h>``` to the top of Plotter.cpp (I don't have a Due to test on)

If that works, i'll push an official fix.

shouryasingh, 2017-03-27:
Dear Devin,

Thanks for your reply. I have been able to compile example program
after adding #include <avr/dtostrf.h> to the top of Plotter.cpp and it
seems to be running in Arduino.

However when I open "ArduinoPlotter_processingListener.exe" it still show blank.

Any ideas regarding that, or some thing which I can do from my side to
make it working.

With regards
Shourya

devinaconley, 2017-03-27:
Hmm that name looks an older version of the listener. Where did you download that from? Check out [this page](https://github.com/devinaconley/arduino-plotter/wiki/Installation-and-Quickstart)

shouryasingh, 2017-03-28:
Dear Devin,

Thanks for your reply.

Now, I am using the latest version of Listner From GitHub site and
file name for 64 Bit windows version is "listener.exe". I get message
on Screen "Scanning Serial Ports...(COM1)"   and "Scanning Serial
Ports...(COM4)"  but again rest of the screen is Blank. My Arduino Due
is installed on COM Port 4.

Also I have Latest Java version 8 on my PC.

Please let me know if there is any way to debug the program running on
Arduino using Serial Monitor/Plotter. Or if there is any other way to
look into this problem.

With regards
Shourya

With regards
Shourya

devinaconley, 2017-03-28:
Can you post the output from the serial monitor? You will have to switch the baud rate to 115200 and cannot have the serial monitor and listener open at the same time.

shouryasingh, 2017-03-28:
Output of serial monitor is also completely blank for Basic Example.

devinaconley, 2017-03-28:
Okay, sounds like the example isn't running at all then. I would verify that everything is compiled (with the fix from above) and uploaded to your arduino

shouryasingh, 2017-03-28:
Yes, every thing compiles without error after including the header
file as told by you and it is uploaded to arduino also. As I have done
some testing (by turning on-board led on-off at different stages of
program) it seems that program enters setup routine but then does not
go in loop and probably hangs somewhere in between.

Also this may be Due specific as you told that program runs well on
other arduino boards.

devinaconley, 2017-03-28:
Yep, it does look like some problem with using dtostrf across different boards. I will write my own function to replace that and put in a fix tonight.

In the meantime, you can try googling for quick fixes. Seems like this is a common problem on the Due.

devinaconley, 2017-03-29:
Pushed a fix for this ([2.2.1](https://github.com/devinaconley/arduino-plotter/releases/tag/2.2.1)). Should show up in the library manager soon.

Let me know if that solves the problem

shouryasingh, 2017-03-30:
Dear Devin,

Thanks for your support. I am sorry for the late reply as I was not
able to use my PC yesterday.

I updated to Latest version of Plotter Library. Now I am able to
compile and upload any example program to Arduino without any
modification.

However, I still get Blank screen on Listner. I have downloaded
Processing 3.3 and tried to run Listner directly from source. It runs
but still I get Blank Screen with same message "Scanning Serial
Ports...."

I also tried another example program (available on internet) to check
communication between Processing and Arduino and it works. It is a
very simple program to display single int incriminating values but it
confirms that communication is ok.

I suspect that problem is somewhere in compatibility of Arduino
Program with Due Board.

With regards
Shourya

devinaconley, 2017-03-30:
Hmm okay. If you open the arduino serial monitor, is it still blank?

shouryasingh, 2017-03-30:
Yes it is completely blank. I seems that program hangs in Arduino Due
during initialization stage and does not enter the loop at all.

devinaconley, 2017-04-01:
Can you try running this?
```arduino
#include "Plotter.h"

double x; // global variables
Plotter p;

void setup()
{
  Serial.begin( 115200 );
  Serial.println( "a" );

  p = Plotter(); // create plotter
  Serial.println( "b" );
  
  p.AddTimeGraph( "Some title of a graph", 500, "label for x", x ); // add any graphs you want
  Serial.println( "c" );
}

void loop()
{
  Serial.println( "d" );
  x = 10*sin( 2.0*PI*( millis() / 5000.0 ) ); // update your variables like usual
  Serial.println( "e" );
  p.Plot(); // plot all current data -- usually called within loop()
  Serial.println( "f" );
}
```

shouryasingh, 2017-04-02:
Dear Devin,

Please find find below the video link for the sample program running on my
PC which was sent by you :-

https://drive.google.com/file/d/0Bw7k2ZCMzF3kNUstQUZvT2hHSktscWdicEFRZDFHRS00OUNz/view?usp=sharing

As you can see there is no output on serial monitor.

With regards
Shourya

devinaconley, 2017-04-07:
Okay, finally got hold of a Due and found the issue. Putting in a fix shortly

devinaconley, 2017-04-07:
Pushed ([2.3.0](https://github.com/devinaconley/arduino-plotter/releases/tag/2.3.0)). Note the change in starting up the plotter in the examples.

Let me know if that works for you.

shouryasingh, 2017-04-09:
Dear Devin,

Thanks for your kind efforts.

I am away from my PC currently for a small family vacation. I will return
in two days and let you know the results.

With regards
Shourya

shouryasingh, 2017-04-11:
Dear Devin,

Yes, now I am getting Plots on Listner.

I would like to Thank your for solving the problem related to Arduino Due Board.

Also if possible please  know the debugging techniques used by you for
Arduino programming. I am quite new to Arduino Programming and work
generally on LabView Platform (which has got extensive debugging
options).

Thanks once again for solving the problem.

With regards
Shourya

devinaconley, 2017-04-11:
Great, glad that's working now. Thanks for reporting this.

Unfortunately, I do not know of any proper debugging solutions for Arduino. I still rely on print statements and the gradual introduction of new code to identify problems.

closed 2017-04-11 by devinaconley