devinaconley/arduino-plotter #14: How to Add arduino plotter sketch with another sketch

jamipraveeen, 2017-03-28:
I want to monitor electrical parameters in real time. I have the following code with emon arduino sketch, I want to plot graph with the data from it usig arduino plotter, How to do that. The following code is below.
`#include <EmonLib.h>

// EmonLibrary examples openenergymonitor.org, Licence GNU GPL V3

#include "EmonLib.h"             // Include Emon Library
EnergyMonitor emon1;             // Create an instance

void setup()
{  
  Serial.begin(250000);
  
  emon1.voltage(2, 10000, 1.7);  // Voltage: input pin, calibration, phase_shift
  emon1.current(1, 500);       // Current: input pin, calibration.
}

void loop()
{
  emon1.calcVI(20,2000);         // Calculate all. No.of half wavelengths (crossings), time-out
  emon1.serialprint();           // Print out all variables (realpower, apparent power, Vrms, Irms, power factor)
  
  float realPower       = emon1.realPower;        //extract Real Power into variable
  float apparentPower   = emon1.apparentPower;    //extract Apparent Power into variable
  float powerFActor     = emon1.powerFactor;      //extract Power Factor into Variable
  float supplyVoltage   = emon1.Vrms;             //extract Vrms into Variable
  float Irms            = emon1.Irms;             //extract Irms into Variable
}`

devinaconley, 2017-03-28:
Check out some of the examples using multiple graphs and multiple variables.

You can't do any printing because the plotter relies on the serial port.

jamipraveeen, 2017-03-29:
No, I want to use this code in your plotter sketch.Can you link both the sketches to get graph?

devinaconley, 2017-03-29:
Yes, you just need to use the energy monitor values as your graphing variables.

jamipraveeen, 2017-03-29:
Yes,but am not getting graph

devinaconley, 2017-03-29:
Are you able to run the examples?

jamipraveeen, 2017-03-29:
Yes, I am able to run your examples but how to integrate with my sketch

jamipraveeen, 2017-03-29:
Please help me out how to integrate my sketch with your plotter sketch... I will add credits in my project

devinaconley, 2017-03-29:
You can try something along these lines (**Note this is untested**):
```arduino
// EmonLibrary examples openenergymonitor.org, Licence GNU GPL V3

#include "EmonLib.h" // Include Emon Library
EnergyMonitor emon1; // Create an instance

#include "Plotter.h"
Plotter p;
float realPower;

void setup()
{
  emon1.voltage(2, 10000, 1.7); // Voltage: input pin, calibration, phase_shift
  emon1.current(1, 500); // Current: input pin, calibration.

  p = Plotter(); // create plotter
  p.AddTimeGraph( "Real power vs time", 500, "real power", realPower ); // add any graphs you want
}

void loop()
{
  emon1.calcVI(20,2000); // Calculate all. No.of half wavelengths (crossings), time-out

  realPower = emon1.realPower; //extract Real Power into variable

  p.Plot();
}
```

jamipraveeen, 2017-03-29:
graph is not visible i mean its not plotting

devinaconley, 2017-03-29:
Can you post the output from the serial monitor? Baud rate needs to be 115200 and listener needs to be closed

devinaconley, 2017-04-11:
Closing this. Feel free to reopen if you still need help.

closed 2017-04-11 by devinaconley