devinaconley/arduino-plotter #15: Can't make listener work

kjeffris, 2017-04-01:
I am trying to view just simple changes in resistance in real time for a presentation I am giving tomorrow, and I can't seem to make this work.  I am using Atom PlatformIO with an arduino 101, and am using the following code:

`//#include "Arduino.h"
#include "Plotter.h"

double rVal = 10e3; // Known resistor value.
int iPin = A2;   // Pin our input is connected to.
double vIn = 5;   // Input voltage corrected for loss.
Plotter p; // Global Plotter variable
double out; // Global output variable
double resistance;

void setup()
{
  // put your setup code here, to run once:
  p = Plotter();          // Initialize Plotter
  p.AddTimeGraph( "Resistance", 500, "Resistance", resistance);

void loop()
{
  // put your main code here, to run repeatedly:
  out = analogRead(iPin);            // Our output voltage
  resistance = (out * vIn) / (vIn - out);  // The resistance between leads
  p.Plot();
}`

However when I open the Listener app it switches off between displaying 
[image]
and
[image].

Any advice?

devinaconley, 2017-04-01:
Are you able to run the examples?

devinaconley, 2017-04-11:
Closing this. Feel free to reopen if you still have a problem.

closed 2017-04-11 by devinaconley