devinaconley/arduino-plotter #20: Invalid messages with too many plots

Chritzel, 2018-07-13:
Hey Devin,

first of all: Thank you for this great and easy to use plotting-system, it makes working with real-time-sensor-evaluation so much easier!

However I recently experienced a small issue (I have a solution already, but let me first motivate):

I'm working on an Arduino Uno. I was trying to plot four TimeGraphs with 4 (or 5) variables each, all complete with title and label descriptions. Everything compiled fine, however nothing was being transmitted (TX LED not lit up).
I commented out to plots, and indeed, now the TX LED was lit up. However, when I started the listener-application, all I got was a gray screen after it connected to the serial port.

I took a look at the transmitted serial data, and as it turned out, some of the data of the description line (the one with the names of the graphs and everything) was dropped. This resulted in a syntax error in the resulting JSON-String, since also a " was dropped.

After some more trying out I found out that this was a memory-error - my small AtMega had simply run out of memory, and so some stuff was just dropped. So this is probably more an optimization request, than a bug-report.

What I did to fix my issue was the following. In Plotter.h and Plotter.cpp:

- I replaced the usage of `String` with `const char *`. Since you were only using String as a container anyways, this was no issue. String needs a bit more memory than const char *.
- I split up the Serial.print()-statements. I had to do this anyway, because concatenation doesn't work that way with const char *. But this would also probably lead to some improvements if you decide to keep String, because the String-concatenation in the print()-calls (using "+") also takes a bunch of memory.
  
  You can do the splitting up by just replacing every occurence of "" + " with ""); Serial.print(" and every occurence of " + "" with "); Serial.print("". 

After those two changes, everything worked again - nothing got dropped, and all plots were displayed like they should.

Let me know what you think, if you want I can also prepare a pull-request. Thanks again for the great library!

devinaconley, 2018-07-21:
Hey @Chritzel - thanks for reporting this and looking into the issue. Those fixes make sense to me. If you could put together a PR that would be great!

devinaconley, 2018-08-03:
thanks! merged #21  -- closing

closed 2018-08-03 by devinaconley