devinaconley/arduino-plotter #28: Can not install

clplaneguy, 2019-04-12:
In the Arduino IDE, I can use the Serial Plotter to see a live display of the variables values as the program runs.  I also use Veusz graph plotting software to graph the data after the run is over.  Veusz is much better at graphing than the Arduino IDE.  Veusz also has a feature to collect data from a file every 0.1 sec (and other).  I would like to be able to do the same thing in PlatformIO.  Is this possible?  I can use an external Terminal program.  But how do I get a live plot.

platformio lib install 850
Does not work.

devinaconley, 2019-04-17:
Please follow the instructions for setup here:
https://github.com/devinaconley/arduino-plotter/wiki/Installation-and-Quickstart

devinaconley, 2019-05-01:
Closing this. Feel free to reopen if there is a more specific problem.

closed 2019-05-01 by devinaconley