devinaconley/arduino-plotter #30: Print current data value along with the legend

vacky11, 2019-07-08:
Is there a way to print current data value (in case of a time graph) along with the legend for each plot?

phdv61, 2019-07-09:
I don’t see any reason why not.

To get continuous curves, I, modified the « graph » component.
I also modified it to get fixed vertical scales,  and be able to display three graphs vertically, instead of getting 2+ 2 as the present software package does.

The easiest way I think, since « Graph » gets the recent value(s) to be displayed, it to modify the display where the test of the legend appears, and replace it by a string built with the value received, and redraw it each time a new value is received.

Or you add a string somewhere and draw the value wanted converted to string , the same way the legend is.

This is what I would do.
Hope it helps.

regards.

Philippe from France.

vacky11, 2019-07-09:
Thanks for replying Philippe

Well, I wanted to know if this is already implemented and if some parameter needs to be enables to print current value. But looks like it is not.

I have modified single line in Graph.java to do this

`	// Draw each legend entry
	for ( int i = 0; i < this.numVars; i++ )
	{
	    this.parent.fill( this.colors[i] );
	    this.parent.text((this.labels[i] + ": " + Double.toString(this.data[this.index-1][i][1])), this.posX + this.width - 10, textPos);
	    textPos += ( labelSz + labelSz/4 );
	    this.parent.stroke( this.colors[i] );
	}`

devinaconley, 2019-08-08:
@vacky11 - this feels like a useful feature. Would welcome a contribution if you open a PR with this enhancement