devinaconley/arduino-plotter #31: No serial output

CafeNoir64, 2019-08-25:
Hi, I hope I'm putting this in the right place.

I am using an ESP32 in platformio and I am trying to use the provided examples (after including the arduino.h library as is required in platformio). I have also installed the listener application (32 bit, because 64 bit version did not seem to work for me) and java runtime environment.

The plotter says scanning for serial ports ... (COM10), so I assume it is working properly and found the only active com port. However, no data is shown at all. Also opening a terminal in platformio does not even show any raw data coming through. The example does compile and flash succesfully to the board. So somehow the board is not generating any serial output.

My platform.ini file is configured the following:
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
monitor_speed = 115200

Another strange thing I encountered is that I cannot even get any Serial.println() commands to work, it only starts working if I remove the Plotter p; statement. 

I already tried different boards but have the same issue, they do all work with different sketches so I doubt its a hardware issue. I also tried using v2.3.0 instead of the latest 2.3.3 version of plotter, but also no luck there.

I am quite lost so I hope someone can point me in the right direction of whats going wrong here. Am I missing any other dependency for this library to function properly?

devinaconley, 2019-08-26:
Hi @CafeNoir64 - the plotter does initialize the serial port with a baud rate of 115200, so you should be able to see serialized json messages as raw output in the console. This is a good first sanity check.

Can you make sure the listener is not open and that nothing else is being done with the serial port outside of the library?

CafeNoir64, 2019-08-26:
Hi Devin, thanks for the quick reply. Yes nothing else is utilizing the serial port and I do get some serial output in the console upon boot of the board. Just nothing after that.

CafeNoir64, 2019-08-28:
I am using the lolin 32 lite board (which has been discontinued now and replaced by the D32)

However, I did a little sanity check first to see if the examples are working in the arduino IDE. And YES they did! So I got to thinking what could actually be different between the two and I saw that the flash frequency is 80MHz in arduino IDE and 40 in platformio by default, also the flash mode is QIO by default in arduino IDE and DIO in platformio. After changing both of these settings it also worked in platformio! Btw I also didn't have any problems with the 64bit listener app as I thought before that it wasn't functioning.

So my .ini file now looks like this:

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino

monitor_speed = 115200
board_build.f_flash = 80000000L
board_build.flash_mode = qio

The weird thing is however, I believe these settings only influence the flashing speed and should not really affect the code itself (I don't have much of a deep understanding about microcontrollers though). So it still kinda bothers me that I don't know why it is working now, but anyway it is.

Thank you Devin for your valuable time and also creating this library!

closed 2019-09-03 by CafeNoir64

devinaconley, 2019-09-04:
Glad you were able to get things running!

wacher74, 2020-03-17:
Hi,
Same issue here. I use esp32doit-devkit-v1. 
Platformio + witty board = success
Platformio + esp32doit-devkit-v1 = nothing happend, there is no serial out. Absolutly nothing.
But CafeNoir64's solution is working here too!