devinaconley/arduino-plotter #33: enhancement / recording data to text file for further analysis

phdv61, 2020-02-04:
Hello @devinaconley 
I modified your "listener" source code attached below to be able, by pressing the space key, to toggle time-stamped recording of all variables on hard disk, in a date&time-stamped CSV - text file. first line gives the names of the variables stored in the file.
rgds

/*
  ===========================================================================================
  This listener is the main processing script that corresponds to the Arduino Plotter 
  library for Arduino. This driver script handles serial port information and manages a 
  set of Graph objects to do the actual plotting. 
  -------------------------------------------------------------------------------------------
  The library stores and handles all relevant graph information and variable references, 
  and transfers information via the serial port to a listener program written with the
  software provided by Processing. No modification is needed to this program; graph placement,
  axis-scaling, etc. are handled automatically. 
  Multiple options for this listener are available including stand-alone applications as well 
  as the source Processing script.

  The library, these listeners, a quick-start guide, documentation, and usage examples are 
  available at:
  
  https://github.com/devinaconley/arduino-plotter

  -------------------------------------------------------------------------------------------
  Arduino Plotter Listener Modified PhDV61 : On-off recording in a text file for excel off-line analysis)
  v2.2.1
  https://github.com/devinaconley/arduino-plotter
  by Devin Conley
  ===========================================================================================
*/

import processing.serial.*;
import java.util.Map;

import java.util.concurrent.Callable;
import java.util.concurrent.ExecutorService;
import java.util.concurrent.Executors;
import java.util.concurrent.Future;
import java.util.concurrent.TimeUnit;
import java.util.concurrent.TimeoutException;

// FLAG FOR DEBUG MODE
final boolean DEBUG = false;

//CONSTANTS
final char OUTER_KEY = '#';
final int  MARGIN_SZ = 20; // between plots
final int  BG_COL = 75; // background
final int  PORT_INTERVAL = 5000; // time to sit on each port
final int  CONNECT_TIMEOUT = 2000; // force timeout on connecting to serial port
final int  BAUD_RATE = 115200;

final HashMap<String, Integer> COLORMAP = new HashMap<String, Integer>()
{
    {
	put( "red", color( 255, 0, 0 ) );
	put( "green", color( 0, 255, 0 ) );
	put( "blue", color( 0, 0, 255 ) );
	put( "orange", color( 255, 153, 51 ) );
	put( "yellow", color( 255, 255, 0 ) );
	put( "pink", color( 255, 51, 204 ) );
	put( "purple", color( 172, 0, 230 ) );
	put( "cyan", color( 0, 255, 255 ) );
    }
};

// Setup and config Globals
int h;
int w;
int numGraphs;
String configCode = "This will not be matched!";
[... 349 lines cut ...]
    {
	return new Serial( this.parent, this.portName, baudRate );
    }     
}

// Logger helper
void logMessage( String message, boolean debugOnly )
{
    if ( DEBUG || !debugOnly )
    {
	String level = debugOnly ? "DEBUG" : "STATUS";
	println( "[Time: " + millis() + " ms]" + "[" + level + "] " + message );
    }
}

void keyPressed() {
  if (key==ESC)  // to exit listener
    {
    output.flush(); // Writes the remaining data to the file
    output.close(); // Finishes the file
    exit(); // Stops the program  
    } 
  else if (key==' ')  // to toggle record - otehr keys can be implemented to do other things
    Recording=!Recording;
}

devinaconley, 2020-04-17:
@phdv61 - would welcome a PR to add optional logging from the listener

devinaconley, 2020-05-20:
Closing as this request is already being tracked here #12

closed 2020-05-20 by devinaconley