devinaconley/arduino-plotter #34: quick_start.ino not working

lasys, 2020-05-27:
Hi, 

i just found your programm - very cool.

Two points:
1.
When i ran the quick_start.ino on an esp32 i had to remove`p.Begin();`. Otherwise it does not print anything out or show in your listener application. 

2. 
When i add some `Serial.println("whatever");` after the plot command it does not work. 
Is this your intention?

Thanks!

_quick_start.ino_
```
/*
  ===========================================================================================
  Example used in Quick-Start
  -------------------------------------------------------------------------------------------
  Plotter
  v2.3.0
  https://github.com/devinaconley/arduino-plotter
  by Devin Conley
  ===========================================================================================
*/

#include "Plotter.h"

double x;

Plotter p;

void setup()
{
    p.Begin();

    p.AddTimeGraph( "Some title of a graph", 1500, "label for x", x );
}

void loop() {
    x = 10*sin( 2.0*PI*( millis() / 5000.0 ) );

    p.Plot(); // usually called within loop()
}
```

prismv, 2020-06-21:
I second this. My comments I made few months ago I used Plotter with ESP32 and arduino framework:

```C
//Serial.begin(115200); 
// ... this is done in both, Plotter::Plotter(), i.e. before setup() call, and in plotter.Begin()
// and blocks esp32 execution at second call
// Is this a bug in Plotter or esp32-arduino?
Plotter plotter;

void setup() {  
  // dirty fix of "backward compatible" call to Serial in Plotter initialization!
  // because otherwise it blocks execution
  Serial.end();
  delay(10);
  // end fix
  plotter.Begin();
  // ...
}
```

mikeperrotta, 2020-06-25:
I also have an issue that seems related. When I run the `quick_start.ino` example on a Feather Express nRF52840 board, everything works as expected. But when I run the same script on a Feather nRF52832 board, nothing is printed to the serial port and the processing script finds nothing when it listens for configuration. However, if I comment out line 36 in Plotter.cpp, everything works as expected on both boards. https://github.com/devinaconley/arduino-plotter/blob/533124ccfe3730185448e27c79bf39229b906431/src/Plotter.cpp#L36

Although `quick_start.ino` creates successful plots for both boards (once line 36 is commented out), the x-axis does scale very differently for both boards (confirmed that nothing else is changed).

devinaconley, 2020-06-28:
Hi @lasys @prismv @mikeperrotta - thanks for looking into this issue. That call was left in there for a fairly old version. Just removed it on `master`

devinaconley, 2020-06-28:
@lasys - yes that is expected behavior, currently need exclusive usage on the serial port

Just added a note to the wiki here:
https://github.com/devinaconley/arduino-plotter/wiki/Troubleshooting#conflict-with-other-serial-communications

devinaconley, 2020-06-28:
Just stamped a new release:
https://github.com/devinaconley/arduino-plotter/releases/tag/2.4.1

Closing this issue. Feel free to reopen if the problem is not resolved.

closed 2020-06-28 by devinaconley