devinaconley/arduino-plotter #35: Plot FFT result data from Arduino to PC screen as live frequency bar graph

andreyrusak2020, 2020-07-21:
Looking for a good working codes for Arduno and Processing sides to support plot of FFT results in live bar graph style (normally 128 dots).

phdv61, 2020-07-21:
Plenty of code available on the net if you search ' Arduino Processing Bar Graph '
ex : 
https://boolscott.wordpress.com/2010/02/04/arduino-processing-analogue-bar-graph-2/

andreyrusak2020, 2020-07-21:
Unfortunately this code is not workable...(((( I spend many hours to repair I do not knew what is the reason... In fact - what I reached... it is working in testing mode only - i.e. step by step taking data from serial buffer and plot in on screen. But when start normally it is cycling... no answer. I guess this is a problem in Arduino side...

devinaconley, 2020-07-21:
Marking as a feature request.

In the meantime, one possible workaround is to create a scatter graph (XY), limit it to display 128 points, and plot each of your frequency components at a different x value. You will need to plot each value one at a time, but you will end up with one point displayed per frequency component. You can even use one graph for magnitude and one graph for phase angle.

andreyrusak2020, 2020-07-21:
Oh!  I found it !!!!! :)It is herehttp://wiki.openmusiclabs.com/wiki/ArduinoFHT YoursAndrey 21.07.2020, 15:43, "Devin A.