devinaconley/arduino-plotter #37: Listener opens just a grey screen

Nicolaslarag20, 2021-04-16:
Hi, just found out your code. It seems awesome! The only issue is that it doesnt work for me.

I'm using a Windows 10 machine with Arduino IDE 1.8.13 and an arduino uno.

At first I thought that I was messing something up on my code, but then i tried using the "quick_start" example and its the same.
The listener exe opens but I can only see the whole window in a light grey color. Tried running it as an admin and nothing changes. 

I unistalled my java and then reinstalled and nothing.

Thanks!

[image]

Badkast, 2021-05-15:
I`m currently having the same problem here, did you find a solution yet?

devinaconley, 2021-09-25:
Are you able to see text in the arduino debugger output?

devinaconley, 2021-09-25:
@Nicolaslarag20  @Badkast  are you able to see text in the arduino debugger output?

phdv61, 2021-09-26:
Go and see issue #38 and the work around I suggested.
Rebuild the project with the 'java embeded' option ticked.
There is an issue with the serial port between processing and java making the app crash if not run on the same PC they were built on.

leCandas, 2022-02-06:
One addition to above
These steps does not work for processing V4 and I guess you don't have listener.java anymore.
Try steps with Processing V3 and ignore listener.java then it works.