devinaconley/arduino-plotter #38: listener.exe crashes immediately

NoahDavid18, 2021-06-10:
When I run the listener.exe it crashes immediately. I downloaded the windows 64 bit stand-alone listener. 
This is  the first part of the log file it makes after crashing: 
```
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x000000007110b5db, pid=10316, tid=0x0000000000003a64
#
# JRE version: Java(TM) SE Runtime Environment (8.0_291-b10) (build 1.8.0_291-b10)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (25.291-b10 mixed mode windows-amd64 compressed oops)
# Problematic frame:
# C  [jSSC-2.8.dll+0xb5db]
#
# Failed to write core dump. Minidumps are not enabled by default on client versions of Windows
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#
```

phdv61, 2021-06-11:
I have experienced many crashes with processing apps when Java was not 'embeded', and when running the generated .exe file on a different PC fitted with the java machine.

This is apparently due to the way processing and java handle the serial port and maybe a problem with the location of some files.

Since then, I always export the app with the 'java embeded' option ticked when I make use of the serial port. And it fixes the crash.

gabomination, 2021-06-24:
Hi, I am experiencing the same exact problem. The answer from phdv61 isn't really clear to me : does it mean the app needs to be rebuilded ?
Until now, I was trying to use the listener precompiled version. I am not familiar to java.

phdv61, 2021-06-24:
Yes, this is what I would suggest.
get  processing from processing.org, install the latest java machine from Oracle on your PC, download all listener project file, and rebuild the listener project / export it with the 'java embeded' option ticked.

There is an issue with running an .exe file on a different PC from the one it was built on, when using the serial port.

I found that work around.

gabomination, 2021-06-25:
Hey, thanks for the very quick answer. I tried it right now, and you were right : it works. Nice !

gpila7, 2021-06-27:
Hi, I'm having the same problem. I downloaded processing and the latest java. I'm confused on what phdv61 mean by rebuild the listener project. And also, where and what should I export with the java embedded option?

Thanking in advance.

gabomination, 2021-06-28:
Hi,
Open the project in the listener dir (.pde file) with processing. Then, file -> export -> check "java embedded".
The app should appear in a build dir. And this time it should work.

devinaconley, 2021-09-25:
Thanks for the report, will generate a new set of listener exe's soon

alexstratford, 2023-03-04:
I'm having the same issue, any news on a fix?