devinaconley/arduino-plotter #42: Code Does Nothing...

Shadow7even, 2021-12-03:
I updated my windows before starting to use Arduino for a project. I'm not sure why, but the code doesn't do anything for me. The code Don't open a window or graph. I wanted to show the difference in voltage/resistance when using a specific sensor and plot a graph. The quick setup still gave me no results. Additionally, when I open the Serial monitor it starts filling lines with symbols such as "��6��q��p@�*", and it goes on, Filling the whole Serial Monitor screen infinitely.

I downloaded the library because I couldn't use the Serial Plotter. The Plotter used to be under Tools on the IDE (built-in). Is there a way to quickly use the info (numbers) received in the Serial Monitor and Plot it in a Windows 11, 2.0.0-Beta12 Arduino IDE?

If anyone knows what to do, I'll appreciate it. If any extra info is needed, I'll be glad to provide it.

[image]

phdv61, 2021-12-03:
The listener code has one component running on Arduino, the other one on your PC ( to be compiled with the 'processing' environment).
Both codes communicate through the serial port and exchange more than just your data as you may imagine. The speed is 115200 if I remember well, so unless you set the baud rate properly, you won't see anything. But the monitor can't be used as you will not have the right exchange of data. you must run one app on the Arduino, and the other on your PC. If your Arduino board is connected through USB, you need, depending upon which kind of PC you use, use the right com port with the right name. Last but not least the listener .exe does not run on Windows 64 bits unless you install java, and you recompile as there are some problems with the java machine and the way it handles the serial port. So better have java installed first, and recompile the PC component.

Shadow7even, 2021-12-06:
> The listener code has one component running on Arduino, the other one on your PC ( to be compiled with the 'processing' environment). Both codes communicate through the serial port and exchange more [...]

I wasn't expecting it to only work on a specific baud. Since they (PC and Arduino) all process faster than 115200 bits/sec and that is mainly the speed that is displayed. Something like fps in a game... you can have 120, 60, or 30. I thought the baud wouldn't matter. But I'll try it out setting it to 115200.

Additionally, I am not sure If I installed Java on my computer. That might be the biggest issue. I get the data normally, 9600 baud works when I'm using the serial monitor. It shows the numbers very fast, but it is possible to have an idea of the data. The graph would help a lot. I will try to install it and recompile it again. Thanks for stopping by and helping me have at least a direction!

phdv61, 2021-12-06:
You can easily edit the code and change the COM speed in the source code.
During the init phase, Arduino and the PC exchange data to make sure they are "authorized" to communicate with each other, and exhange the required "graphic settings". it comes as a "package".
You can develop your own code to exchange data between Arduino and your PC, sending simple ascii/string messages like
STX, "your data1, your data2, ....", ETX from Arduino
and receiving/decoding them on your own PC graphic software.